astsam/rtl8812au #98: Compilation problem Raspberry Pi 3 Kernel 4.14

movitch, 2018-09-12:
Hi,
I use your driver on Rasperry pi 3 with Raspbain Jessie and linux Kernel 4.9.
It works very fine and i'm very satisfated to use it.
I'm obliged to change and using Pi3B+ with stretch and linux Kernel 4.14.
I have compilation problem in ioctl_cfg80211.o.
I readed your driver was not compatible with this kernel. Is there any way to solve this compilation problem ? By using a patch ?
I tried some other repository but i wish to continue using your driver because RPI3 experience is very positive.
Thanks by advance for your help

kimocoder, 2018-09-12:
Which branch? which error? needs logs here (compile log/output)

movitch, 2018-09-13:
Sorry for missing informations.
- Branch : 5.1.5
Compilation launched with "make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm" (same as pi3 kernel 4.9)
- Compilation log :

make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /lib/modules/4.14.62-v7+/build M=/opt/cora/astsam/rtl8812au  modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.14.62-v7+ »
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_cmd.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_security.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_debug.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_io.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_mlme.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_mi.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_vht.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_rf.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_recv.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_ap.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_xmit.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_p2p.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_tdls.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_br_ext.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_iol.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_sreset.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_btcoex.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_beamforming.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/rtw_odm.o
  CC [M]  /opt/cora/astsam/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /opt/cora/astsam/rtl8812au/os_dep/osdep_service.o
  CC [M]  /opt/cora/astsam/rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /opt/cora/astsam/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /opt/cora/astsam/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /opt/cora/astsam/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /opt/cora/astsam/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /opt/cora/astsam/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.o
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c:759:6: error: passing argument 2 of ‘cfg80211_roamed’ from incompatible pointer type [-Werror=incompatible-pointer-types]
    , notify_channel
      ^~~~~~~~~~~~~~
In file included from /opt/cora/astsam/rtl8812au/include/osdep_service_linux.h:91:0,
                 from /opt/cora/astsam/rtl8812au/include/osdep_service.h:41,
                 from /opt/cora/astsam/rtl8812au/include/drv_types.h:32,
                 from /opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5479:6: note: expected ‘struct cfg80211_roam_info *’ but argument is of type ‘struct ieee80211_channel *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c:761:6: warning: passing argument 3 of ‘cfg80211_roamed’ makes integer from pointer without a cast [-Wint-conversion]
    , cur_network->network.MacAddress
      ^~~~~~~~~~~
In file included from /opt/cora/astsam/rtl8812au/include/osdep_service_linux.h:91:0,
                 from /opt/cora/astsam/rtl8812au/include/osdep_service.h:41,
                 from /opt/cora/astsam/rtl8812au/include/drv_types.h:32,
                 from /opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5479:6: note: expected ‘gfp_t {aka unsigned int}’ but argument is of type ‘unsigned char *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c:757:3: error: too many arguments to function ‘cfg80211_roamed’
   cfg80211_roamed(padapter->pnetdev
   ^~~~~~~~~~~~~~~
In file included from /opt/cora/astsam/rtl8812au/include/osdep_service_linux.h:91:0,
                 from /opt/cora/astsam/rtl8812au/include/osdep_service.h:41,
                 from /opt/cora/astsam/rtl8812au/include/drv_types.h:32,
[... 7 lines cut ...]
          ^~
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_preinit_wiphy’:
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6334:18: error: ‘WIPHY_FLAG_SUPPORTS_SCHED_SCAN’ undeclared (first use in this function)
  wiphy->flags |= WIPHY_FLAG_SUPPORTS_SCHED_SCAN;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6334:18: note: each undeclared identifier is reported only once for each function it appears in
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c: At top level:
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6367:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = cfg80211_rtw_change_iface,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6367:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
[... cut ...]
scripts/Makefile.build:328 : la recette pour la cible « /opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.o » a échouée
make[2]: *** [/opt/cora/astsam/rtl8812au/os_dep/linux/ioctl_cfg80211.o] Erreur 1
Makefile:1523 : la recette pour la cible « _module_/opt/cora/astsam/rtl8812au » a échouée
make[1]: *** [_module_/opt/cora/astsam/rtl8812au] Erreur 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.14.62-v7+ »
Makefile:1855 : la recette pour la cible « modules » a échouée
make: *** [modules] Erreur 2 

Thank you