astsam/rtl8812au #105: Doesnt MAKE in kernel 4.15.0-47-generic(Ubuntu 18.04)

juanlet, 2019-04-15:
I'm getting this error in ubuntu 18.04, any ideas on how to fix it?

```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.0-47-generic/build M=/home/home/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-47-generic'
  CC [M]  /home/home/rtl8812au/core/rtw_cmd.o
In file included from /home/home/rtl8812au/include/osdep_service.h:41:0,
                 from /home/home/rtl8812au/include/drv_types.h:32,
                 from /home/home/rtl8812au/core/rtw_cmd.c:22:
/home/home/rtl8812au/include/osdep_service_linux.h: In function ‘_init_timer’:
/home/home/rtl8812au/include/osdep_service_linux.h:273:8: error: ‘_timer {aka struct timer_list}’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;
        ^~
/home/home/rtl8812au/include/osdep_service_linux.h:274:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer
In file included from /home/home/rtl8812au/include/drv_types.h:32:0,
                 from /home/home/rtl8812au/core/rtw_cmd.c:22:
/home/home/rtl8812au/include/osdep_service.h: In function ‘thread_enter’:
/home/home/rtl8812au/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/home/home/rtl8812au/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/home/rtl8812au/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
      timer_pending
/home/home/rtl8812au/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: some warnings being treated as errors
scripts/Makefile.build:332: recipe for target '/home/home/rtl8812au/core/rtw_cmd.o' failed
make[2]: *** [/home/home/rtl8812au/core/rtw_cmd.o] Error 1
Makefile:1552: recipe for target '_module_/home/home/rtl8812au' failed
make[1]: *** [_module_/home/home/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-47-generic'
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2

```

marc-y-marc, 2019-06-12:
Same issue here

dontdeadopeninside, 2019-08-04:
Got the same problem. kernel 4.19 Kali linux

starhawk64, 2022-07-12:
Same issue now in mid-July 2022, Ubuntu 18.04. I have a specific reason for using this edition of Ubuntu on the hardware I am using; this error and the fact that I am pressed for time now mandates a configuration change to the hardware in question, which is a non-trivial task at best... especially since variants of this issue are present on upstream forks as well, such as the variant offered by 'aircrack-ng'.

More importantly, why is this bug still extant on a piece of software that appears to be actively maintained, over _three years_ from initial report...?! Not everyone can "just use newer software" or "just use a newer platform" -- Linux may be free, as a kernel proper and bundled within an OS that has been built around it, sure, but the hardware on which it runs is not... and sloppy maintenance of a package or otherwise-bundled-or-assembled software item is in its own way _worse_ than no maintenance at all, as it typically (as here) gives the impression of properly-maintained software when it is anything but.

Good thing this isn't a security issue, I bet Johnathan Bennett over at Hackaday would devote an entire _article_ to you at this point, if it was.

Anything worth doing, is worth doing well. Devs, you have a bug to squash. Fix it like your name's Felix already!