astsam/rtl8812au #108: error on make, Manjaro

T2brozz, 2020-02-27:
Hello
I have issues to install the driver to use my alfa awus036ac with monitor Mode.
I use Manjaro (Kernel 5.52-1), Xfce 4.
If I want to "make" i get this error: 

$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.5.2-1-MANJARO/build M=/home/yannes/rtl8812au  modules
make[1]: Entering directory '/usr/lib/modules/5.5.2-1-MANJARO/build'
  CC [M]  /home/yannes/rtl8812au/core/rtw_cmd.o
In file included from /home/yannes/rtl8812au/include/osdep_service.h:41,
                 from /home/yannes/rtl8812au/include/drv_types.h:32,
                 from /home/yannes/rtl8812au/core/rtw_cmd.c:22:
/home/yannes/rtl8812au/include/osdep_service_linux.h: In function ‘_init_timer’:
/home/yannes/rtl8812au/include/osdep_service_linux.h:273:8: error: ‘_timer’ {aka ‘struct timer_list’} has no member named ‘data’
  273 |  ptimer->data = (unsigned long)cntx;
      |        ^~
/home/yannes/rtl8812au/include/osdep_service_linux.h:274:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  274 |  init_timer(ptimer);
      |  ^~~~~~~~~~
      |  _init_timer
In file included from /home/yannes/rtl8812au/include/drv_types.h:35,
                 from /home/yannes/rtl8812au/core/rtw_cmd.c:22:
/home/yannes/rtl8812au/include/wifi.h: At top level:
/home/yannes/rtl8812au/include/wifi.h:1009: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1009 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from ./include/net/cfg80211.h:21,
                 from /home/yannes/rtl8812au/include/osdep_service_linux.h:87,
                 from /home/yannes/rtl8812au/include/osdep_service.h:41,
                 from /home/yannes/rtl8812au/include/drv_types.h:32,
                 from /home/yannes/rtl8812au/core/rtw_cmd.c:22:
./include/linux/ieee80211.h:1451: note: this is the location of the previous definition
 1451 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:266: /home/yannes/rtl8812au/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1693: /home/yannes/rtl8812au] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.5.2-1-MANJARO/build'
make: *** [Makefile:1264: modules] Error 2

I hope somebody can help me :)

MrCobblestoneIV, 2020-04-02:
Same... I have the exact same problem.

MrCobblestoneIV, 2020-04-04:
I've finally worked it out! These are the steps I used:

git clone https://github.com/aircrack-ng/rtl8812au.git
apt-get install dkms
cd rtl*
cd rtl*
./dkms-install.sh
apt-get update
sudo apt-get install build-essential bc libelf-dev linux-headers-`uname -r`
reboot now

I did this and after the reboot I managed to put the Alfa Awus036ac in monitor mode.

coderscoding, 2022-10-06:
> **apt-get** install dkms
> **apt-get** update
> sudo **apt-get** install build-essential bc libelf-dev linux-headers-uname -r

But This Is Not For Arch Linux...

closed 2022-11-26 by T2brozz