astsam/rtl8812au #125: Error 2 during make

ppds07, 2024-07-06:
Here's the error that has returned while using the command "$ make" after cloning the git.

pi5@raspberrypi:~/rtl8812au $ make RTL8811=1 DEBUG=1
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/6.6.31+rpt-rpi-2712/build M=/home/pi5/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.31+rpt-rpi-2712'
  CC [M]  /home/pi5/rtl8812au/core/rtw_cmd.o
In file included from /home/pi5/rtl8812au/include/osdep_service.h:41,
                 from /home/pi5/rtl8812au/include/drv_types.h:32,
                 from /home/pi5/rtl8812au/core/rtw_cmd.c:22:
/home/pi5/rtl8812au/include/osdep_service_linux.h: In function ‘_init_timer’:
/home/pi5/rtl8812au/include/osdep_service_linux.h:273:15: error: ‘_timer’ {aka ‘struct timer_list’} has no member named ‘data’
  273 |         ptimer->data = (unsigned long)cntx;
      |               ^~
/home/pi5/rtl8812au/include/osdep_service_linux.h:274:9: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  274 |         init_timer(ptimer);
      |         ^~~~~~~~~~
      |         _init_timer
/home/pi5/rtl8812au/core/rtw_cmd.c: In function ‘rtw_cmd_thread’:
/home/pi5/rtl8812au/include/osdep_service_linux.h:157:31: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
  157 |         #define thread_exit() complete_and_exit(NULL, 0)
      |                               ^~~~~~~~~~~~~~~~~
/home/pi5/rtl8812au/core/rtw_cmd.c:735:9: note: in expansion of macro ‘thread_exit’
  735 |         thread_exit();
      |         ^~~~~~~~~~~
/home/pi5/rtl8812au/core/rtw_cmd.c:737:1: error: control reaches end of non-void function [-Werror=return-type]
  737 | }
      | ^
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.6.31+rpt-common-rpi/scripts/Makefile.build:248: /home/pi5/rtl8812au/core/rtw_cmd.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.31+rpt-common-rpi/Makefile:1938: /home/pi5/rtl8812au] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.31+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.31+rpt-rpi-2712'
make: *** [Makefile:1264: modules] Error 2

surcouf-git, 2024-08-15:
Did you find already ?

ppds07, 2024-10-15:
Sorry for the late response,
It seems that my raspberry pi wasn't setup properly and complete reset fixed it.

closed 2024-10-15 by ppds07